adafruit/Adafruit_Python_ADS1x15 #1: Can't install with pip

maximedupre, 2016-05-19:
I want to install this lib for Python 3, so I tried `pip3 install --user adafruit-ads1x15`, but it didn't work. I also tried with `pip install --user adafruit-ads1x15`, but it doesn't work either.

First I got this error:

    Collecting adafruit-ads1x15
    Using cached Adafruit_ADS1x15-1.0.1.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-rrh89a59/adafruit-ads1x15/setup.py", line 1, in <module>
        from ez_setup import use_setuptools
    ImportError: No module named 'ez_setup'
    
    ----------------------------------------
    Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-rrh89a59/adafruit-ads1x15/

So I installed `ez-setup` with `pip`, but then I got this error:

    Collecting adafruit-ads1x15
    Using cached Adafruit_ADS1x15-1.0.1.tar.gz
    Complete output from command python setup.py egg_info:
    Downloading http://pypi.python.org/packages/source/d/distribute/distribute-0.6.14.tar.gz
    Extracting in /tmp/tmpxysjf7rj
    Traceback (most recent call last):
      File "/home/pi/.local/lib/python3.5/site-packages/ez_setup.py", line 143, in use_setuptools
        raise ImportError
    ImportError
    
    During handling of the above exception, another exception occurred:
    
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-ozskf5n2/adafruit-ads1x15/setup.py", line 2, in <module>
        use_setuptools()
      File "/home/pi/.local/lib/python3.5/site-packages/ez_setup.py", line 145, in use_setuptools
        return _do_download(version, download_base, to_dir, download_delay)
      File "/home/pi/.local/lib/python3.5/site-packages/ez_setup.py", line 125, in _do_download
        _build_egg(egg, tarball, to_dir)
      File "/home/pi/.local/lib/python3.5/site-packages/ez_setup.py", line 99, in _build_egg
        _extractall(tar)
      File "/home/pi/.local/lib/python3.5/site-packages/ez_setup.py", line 467, in _extractall
        self.chown(tarinfo, dirpath)
    TypeError: chown() missing 1 required positional argument: 'numeric_owner'
    
    ----------------------------------------
    Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-ozskf5n2/adafruit-ads1x15/

mecworks, 2016-09-13:
I am seeing the same issues with all the Adafruit python libraries on the Raspberry Pi running the latest Raspbian.  The problem is that my team is moving all our automation code to Python3 and we rely heavily on Adafruit libraries. 

This is the output of an attempted install:

```
# pip3 install --upgrade adafruit-ads1x15
Collecting adafruit-ads1x15
  Downloading Adafruit_ADS1x15-1.0.1.tar.gz
    Complete output from command python setup.py egg_info:
    Downloading http://pypi.python.org/packages/source/d/distribute/distribute-0.6.14.tar.gz
    Extracting in /tmp/tmpo72f3xu8
    Now working in /tmp/tmpo72f3xu8/distribute-0.6.14
    Building a Distribute egg in /tmp/pip-build-zjkkd_wk/adafruit-ads1x15
    creating build
    creating build/src
    creating build/src/docs
    creating build/src/docs/_templates
    creating build/src/docs/_theme
    creating build/src/docs/_theme/nature
    creating build/src/docs/_theme/nature/static
    creating build/src/docs/build
    creating build/src/docs/build/html
    creating build/src/docs/build/html/_sources
    creating build/src/docs/build/html/_static
    creating build/src/setuptools
    creating build/src/setuptools/command
    creating build/src/setuptools/tests
    creating build/src/setuptools/tests/indexes
    creating build/src/setuptools/tests/indexes/test_links_priority
    creating build/src/setuptools/tests/indexes/test_links_priority/simple
    creating build/src/setuptools/tests/indexes/test_links_priority/simple/foobar
    creating build/src/tests
    creating build/src/tests/shlib_test
    copying setuptools/__init__.py -> build/src/setuptools
    copying setuptools/extension.py -> build/src/setuptools
    copying setuptools/archive_util.py -> build/src/setuptools
    copying setuptools/package_index.py -> build/src/setuptools
    copying setuptools/depends.py -> build/src/setuptools
    copying setuptools/sandbox.py -> build/src/setuptools
    copying setuptools/dist.py -> build/src/setuptools
    copying setuptools/tests/__init__.py -> build/src/setuptools/tests
    copying setuptools/tests/test_develop.py -> build/src/setuptools/tests
    copying setuptools/tests/server.py -> build/src/setuptools/tests
    copying setuptools/tests/test_build_ext.py -> build/src/setuptools/tests
    copying setuptools/tests/doctest.py -> build/src/setuptools/tests
    copying setuptools/tests/test_sandbox.py -> build/src/setuptools/tests
    copying setuptools/tests/test_upload_docs.py -> build/src/setuptools/tests
    copying setuptools/tests/test_packageindex.py -> build/src/setuptools/tests
    copying setuptools/tests/test_easy_install.py -> build/src/setuptools/tests
    copying setuptools/tests/test_resources.py -> build/src/setuptools/tests
    copying setuptools/command/saveopts.py -> build/src/setuptools/command
    copying setuptools/command/__init__.py -> build/src/setuptools/command
    copying setuptools/command/easy_install.py -> build/src/setuptools/command
    copying setuptools/command/egg_info.py -> build/src/setuptools/command
    copying setuptools/command/build_py.py -> build/src/setuptools/command
    copying setuptools/command/rotate.py -> build/src/setuptools/command
    copying setuptools/command/upload.py -> build/src/setuptools/command
    copying setuptools/command/upload_docs.py -> build/src/setuptools/command
    copying setuptools/command/build_ext.py -> build/src/setuptools/command
    copying setuptools/command/bdist_wininst.py -> build/src/setuptools/command
    copying setuptools/command/install_lib.py -> build/src/setuptools/command
    copying setuptools/command/sdist.py -> build/src/setuptools/command
    copying setuptools/command/register.py -> build/src/setuptools/command
    copying setuptools/command/setopt.py -> build/src/setuptools/command
    copying setuptools/command/install.py -> build/src/setuptools/command
    copying setuptools/command/develop.py -> build/src/setuptools/command
    copying setuptools/command/test.py -> build/src/setuptools/command
    copying setuptools/command/bdist_egg.py -> build/src/setuptools/command
    copying setuptools/command/alias.py -> build/src/setuptools/command
    copying setuptools/command/bdist_rpm.py -> build/src/setuptools/command
[... 220 lines cut ...]
    
    error: invalid command 'bdist_egg'
    /tmp/pip-build-zjkkd_wk/adafruit-ads1x15/distribute-0.6.14-py3.4.egg
    Traceback (most recent call last):
      File "/usr/local/lib/python3.4/dist-packages/ez_setup.py", line 143, in use_setuptools
        raise ImportError
    ImportError
    
    During handling of the above exception, another exception occurred:
    
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-zjkkd_wk/adafruit-ads1x15/setup.py", line 2, in <module>
        use_setuptools()
      File "/usr/local/lib/python3.4/dist-packages/ez_setup.py", line 145, in use_setuptools
        return _do_download(version, download_base, to_dir, download_delay)
      File "/usr/local/lib/python3.4/dist-packages/ez_setup.py", line 125, in _do_download
        _build_egg(egg, tarball, to_dir)
      File "/usr/local/lib/python3.4/dist-packages/ez_setup.py", line 116, in _build_egg
        raise IOError('Could not build the egg.')
    OSError: Could not build the egg.
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-zjkkd_wk/adafruit-ads1x15/
```

jcaberio, 2016-09-24:
@mecworks you can install it from source, just clone the repo and run

> python3 setup.py install

mecworks, 2016-09-26:
That's what I ended up doing but it would be nice if 'pip install' worked correctly.

dwbro1, 2017-01-15:
Same here.. Here is my output. Running on Pi3 with Raspi lite.

`pi@LaundrySensorsPi:~ $ sudo pip3 install Adafruit_Python_DHT
Downloading/unpacking Adafruit-Python-DHT
  Downloading Adafruit_Python_DHT-1.1.2.tar.gz
  Running setup.py (path:/tmp/pip-build-2b8f4719/Adafruit-Python-DHT/setup.py) egg_info for package Adafruit-Python-DHT
    Traceback (most recent call last):
      File "<string>", line 17, in <module>
      File "/tmp/pip-build-2b8f4719/Adafruit-Python-DHT/setup.py", line 5, in <module>
        import Adafruit_DHT.platform_detect as platform_detect
      File "/tmp/pip-build-2b8f4719/Adafruit-Python-DHT/Adafruit_DHT/__init__.py", line 21, in <module>
        from .common import DHT11, DHT22, AM2302, read, read_retry
      File "/tmp/pip-build-2b8f4719/Adafruit-Python-DHT/Adafruit_DHT/common.py", line 23, in <module>
        import platform_detect
    ImportError: No module named 'platform_detect'
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):

  File "<string>", line 17, in <module>

  File "/tmp/pip-build-2b8f4719/Adafruit-Python-DHT/setup.py", line 5, in <module>

    import Adafruit_DHT.platform_detect as platform_detect

  File "/tmp/pip-build-2b8f4719/Adafruit-Python-DHT/Adafruit_DHT/__init__.py", line 21, in <module>

    from .common import DHT11, DHT22, AM2302, read, read_retry

  File "/tmp/pip-build-2b8f4719/Adafruit-Python-DHT/Adafruit_DHT/common.py", line 23, in <module>

    import platform_detect

ImportError: No module named 'platform_detect'

----------------------------------------
Cleaning up...
Command python setup.py egg_info failed with error code 1 in /tmp/pip-build-2b8f4719/Adafruit-Python-DHT
Storing debug log for failure in /root/.pip/pip.log`

kgspears, 2017-01-23:
I am running RPi3, PIXEL, update 1/8/2017.  The reco setup worked for installing python 2.7, but it seemed lengthy.  Then the Python 3.4 version was installed by sudo pip3 install adafruit-ads1x15.  I used sudo pip freeze > listPy2.7 packages.txt  and sudo pip3 freeze > listPy3.4 packages.txt and comparing to prior outputs the same 3 packages were now in both 2.7 and 3.4, which has unique paths for 2.7 and 3.4.  Both Idle2, Idle3, Spyder (which is 2.7) and Spyder3  all could run the sample code.  The main directory is in home/pi.  Since these drivers are "packaged" future code should have fewer path issues, I hope.  Recommend Spyder/Spyder3 for debugging Python code.  They are accessible from the gui provided in Pixel as preconfigured packages; I installed all six items that show up on search.
   The board was a Grove16BitADC.  They do not have drivers for Python 3.  The Grove 2.7 drivers are stand-alone and based on older Adafruit drivers with extra comments and one error fix.  That might have advantages of documentation and tweeking.  The forum:  http://www.switchdoc.com/forums/topic/would-like-a-python-3-4-version-for-ads1115 discusses how to set up 2.7 drivers to be uniquely called and not confused with other ads1115 drivers having the same names...which can be a problem.   
    The author should fix the readme file to note that his recommended method only installs v 2.7 and that the indicated pip3 command can install to python 3.4.

caternuson, 2018-12-07:
Hi! We are deprecating this library. This issue is either fixed or may need updating for the new library:
https://github.com/adafruit/Adafruit_CircuitPython_ADS1x15

Please see issue summary here:
https://github.com/adafruit/Adafruit_Python_ADS1x15/issues/13

closed 2018-12-07 by caternuson