adafruit/Adafruit_Python_ADS1x15 #4: Raspberry Pi 3 ADS1115 - Remote I/O error

jonathanrjpereira, 2017-10-15:
Thank you for opening an issue on an Adafruit Python library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

- **Do not use GitHub issues for troubleshooting projects and issues.**  Instead use
  the forums at http://forums.adafruit.com to ask questions and troubleshoot why
  something isn't working as expected.  In many cases the problem is a common issue
  that you will more quickly receive help from the forum community.  GitHub issues
  are meant for known defects in the code.  If you don't know if there is a defect
  in the code then start with troubleshooting on the forum first.

- **If following a tutorial or guide be sure you didn't miss a step.** Carefully
  check all of the steps and commands to run have been followed.  Consult the
  forum if you're unsure or have questions about steps in a guide/tutorial.

- **For Python/Raspberry Pi projects check these very common issues to ensure they don't apply**:

  - If you are receiving an **ImportError: No module named...** error then a
    library the code depends on is not installed.  Check the tutorial/guide or
    README to ensure you have installed the necessary libraries.  Usually the
    missing library can be installed with the `pip` tool, but check the tutorial/guide
    for the exact command.  

  - **Be sure you are supplying adequate power to the board.**  Check the specs of
    your board and power in an external power supply.  In many cases just
    plugging a board into your computer is not enough to power it and other
    peripherals.

  - **Double check all soldering joints and connections.**  Flakey connections
    cause many mysterious problems.  See the [guide to excellent soldering](https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools) for examples of good solder joints.

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  **Raspberry Pi 3 with Raspbian operating system**

- Python version (run `python -version` or `python3 -version`):  **Python 2.7.13**

- Error message you are receiving, including any Python exception traces:
```
pi@raspberrypi:~/Adafruit_Python_ADS1x15/examples $ python simpletest.py
Reading ADS1x15 values, press Ctrl-C to quit...
|      0 |      1 |      2 |      3 |
-------------------------------------
Traceback (most recent call last):
  File "simpletest.py", line 42, in <module>
    values[i] = adc.read_adc(i, gain=GAIN)
  File "build/bdist.linux-armv7l/egg/Adafruit_ADS1x15/ADS1x15.py", line 192, in read_adc
  File "build/bdist.linux-armv7l/egg/Adafruit_ADS1x15/ADS1x15.py", line 128, in _read
  File "build/bdist.linux-armv7l/egg/Adafruit_GPIO/I2C.py", line 128, in writeList
  File "build/bdist.linux-armv7l/egg/Adafruit_PureIO/smbus.py", line 274, in write_i2c_block_data
IOError: [Errno 121] Remote I/O error
```

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): 

```
To install the library from source (recommended) run the following commands on a Raspberry Pi or other Debian-based OS system:

sudo apt-get install git build-essential python-dev
cd ~
git clone https://github.com/adafruit/Adafruit_Python_ADS1x15.git
cd Adafruit_Python_ADS1x15
sudo python setup.py install
```

jonathanrjpereira, 2017-10-15:
If I run the following command: ` python3 continuous.py`
I get another error: 
`Traceback (most recent call last):
  File "continuous.py", line 7, in <module>
    import Adafruit_ADS1x15
ImportError: No module named 'Adafruit_ADS1x15'
`

jonathanrjpereira, 2017-10-15:
I finally managed to fix this. Turns out it was faulty jumper wires between the ADC and RasPi. They were fine when I tested them with my multimeter continuity test( I guess it was probably the manner in which I held them while testing the continuity gave me a false positive). ALWAYS check your wires. Once I changed the jumper wires, it finally showed up as an i2c address and the code worked. Unfortunately the code still doesn't  work in Python 3.6. I had to run it in Python 2.7.9.
In addition to this, the raw value you get from the ADC is pretty much no use. At the end of the day, when I use an ADC I want my example code to return the '**voltage**' instead of the raw value. I still haven't found a way of converting the raw value into a voltage level.

closed 2017-10-29 by jonathanrjpereira

atlasRand, 2018-04-09:
Concerning the conversion from raw data into a voltage level can you not take a ratio (e.g. 12 bit is 4096) to convert, it works for a DAC but I don't see any reason why it won't work for an ADC (reversed)

jimlynnjulian, 2018-06-24:
The data for conversion is in the documerntation for the chip. 
As a last resort, apply a known voltage to an input, read the result. Divide the voltage by the readout and come up with a multiplication factor. Change the known voltage and multiply by the multiplication factor and examine the result for accuracy and precision.

jonathanrjpereira, 2018-06-24:
Like I said I created a PR with an example which covers this(Voltage Conversion) a few months ago. But it still hasn't been merged yet.

jimlynnjulian, 2018-06-24:
Acknowledged.
Another source of that particular error is chip addressing. Connect the Addr to 5VDC, Vdd, SCL, or SDA and the same error will show up. As a matter of fact, the i2cdetect command is a good test of the address connection.
I've often had problems with cheap 'Dupont' jumpers. I'm goint to try a more expensive set hyped a lot by the company that makes them. If that does not work I'm dumping jumpers altogether moving to a different method.

jimlynnjulian, 2018-06-24:
Have you ever, successfully, addressed, two or more, ADS1115 modules/breakout boards?

jonathanrjpereira, 2018-06-25:
No. So far I have only required single ended measurements. And the 4 inputs have been sufficient for my requirements.