adafruit/Adafruit_Python_ADS1x15 #12: Sample time delay insufficient for normal tolerance

stur0067, 2018-11-18:
The time delay between i2c write and read operations is insufficient to cover the natural part to part variation of ADS1x15 devices. The consequence of this is that some parts return the previous reading whenever queried. This issue only becomes apparent if the gain value changes between two conversions as the value returned will be correct for the previous gain setting.

Currently a delay of (1.0/data_rate + 0.0001) is used.  The datasheet for the ADC states that the sample rate varies up to +/-10%. So, to avoid this potential issue the delay should be changed to (1.1/data_rate)

stur0067, 2018-11-21:
> hiya - we'll be deprecating this library soon and replacing it fully with
> the circuitpython version which will be supported (we're waiting on one
> more breaking interface change before archiving this library for good)
> can you check out
> https://github.com/adafruit/Adafruit_CircuitPython_ADS1x15 and see if
> that does the right thing?

This new library's implementation does properly handle the part to part
conversion time variation as it actually queries the device to see when the
conversion is complete.
It does have a little bit of room for improvement though as the _read
function waits in increments of 10ms if the _conversion_complete query
immediately after the _write_register call returns false. I haven't tested
it but I would guess that for all but the very fastest data rates on the
ADS1015 this first check will always come back indicating incomplete,
forcing a wait time before that next check that is excessive for all but
the lowest data rate setting of the slower ADS1115. This penalty will
severely limit the user's ability to make multiple rapid single conversion,
likely 50-100 samples/second, even when they believe they are using a much
faster data rate.

I would recommend some modifications that changed depending on how closely
you'd like to complete relative to the ADC's conversion completion.
1A. Add a sleep of 1.0/data_rate, the nominal conversion time, before the
while loop checks for conversion completion
2A. Add a sleep of 1.0/(data_rate*1.1)), the minimum conversion time,
before the while loop

Followed by a change to the sleep time within the loop:
1B. A sleep of 1.0/(data_rate*0.10) should loop at most once for case 1A or
twice for 2A above and adds minimal i2c traffic but fully covers the -10%
sampling rate condition
2B. A sleep of 1.0/(data_rate*0.01) should loop at most 10 times for A1 and
20 times for 2A. This will let you more closely catch the ADCs completion

For 2B you could even monitor the average number of additional iterations
the specific device requires and add that much extra time to the sleep
before the while loop.  This will reduce the i2c traffic.

ladyada, 2018-11-21:
ooh do you mind either submitting a PR or adding an issue to that repo? that way we will see it next time we do a sweep thru the librayr! it would be appreciated

stur0067, 2018-11-21:
Sure :)

caternuson, 2018-12-07:
Hi! We are deprecating this library. This issue is either fixed or may need updating for the new library:
https://github.com/adafruit/Adafruit_CircuitPython_ADS1x15

Please see issue summary here:
https://github.com/adafruit/Adafruit_Python_ADS1x15/issues/13

closed 2018-12-07 by caternuson