adafruit/Adafruit_Python_ADS1x15 #13: This library should be deprecated.

caternuson, 2018-12-03:
To be replaced by [Adafruit_CircuitPython_ADS1x15](https://github.com/adafruit/Adafruit_CircuitPython_ADS1x15) and [Blinka](https://github.com/adafruit/Adafruit_Blinka).

For historical reference, my take on the outstanding Issues and PRs:

**Issues:**
* *Sample time delay insufficient for normal tolerance* - fixed in new lib
* *working with threads* - might reappear in new form on blinka boards
* *PyPI improvements* - fixed with new process
* *ADS1x15_CONFIG_GAIN gain=2/3 gives unexpected results* - should be OK in Python3
* *Comparator mode doesn't work for ADS1015* - feature not in new lib yet
* *Error in Intel Edison* - blinka would need to add support
* *Can't install with pip* - fixed with new process

**PRs:**
* *Single Ended Input Example & README Update* - fixed in new lib

ladyada, 2018-12-03:
thanks - yep sounds good to close all issues/PRs and point people to new library!

caternuson, 2018-12-07:
Nothing to see here but this cute ASCII cat....
```
                                 _
                                | \
                                | |
                                | |
           |\                   | |
          /, ~\                / /
         X     `-.....-------./ /
          ~-. ~  ~              |
             \             /    |
              \  /_     ___\   /
              | /\ ~~~~~   \ |
              | | \        || |
              | |\ \       || )
             (_/ (_/      ((_/
```

closed 2018-12-07 by caternuson